woopstar/hsem #14: Energy average sensors lose their queue

woopstar, 2024-10-18:
The energy average sensors are currently loosing their sample queue to calculate on when HA is restarted. This results in wrong calculations

Extati, 2024-11-08:
> The energy average sensors are currently loosing their sample queue to calculate on when HA is restarted. This results in wrong calculations

Will there be an update for this, or do it has little to no meaning?

woopstar, 2024-11-09:
> > The energy average sensors are currently loosing their sample queue to calculate on when HA is restarted. This results in wrong calculations
> 
> 
> 
> Will there be an update for this, or do it has little to no meaning?

Yes. I am currently working on a fix for this. Will keep you posted. 

Also working on adding more sensor information so I can start calculations etc.

woopstar, 2024-11-10:
> Nice, sounds great, really appreciate this is an integration, makes it more easier for the rest of us.
> 
> Great whit more sensor information, one big wish, would be to have it buy on the lowest price from 0-06 where it now just fills up the battery straight at 0 - when it knows there Is no enough power to cover the morning.
> 
> 
> 
> Else great program you have made here - really appreciate it. Looking forward to more futures.
> 
> 
> 
> BR

Currently almost done. The PR is open. Just testing it personally now. 

It will charge the battery to grid ac soc between 0 and 06 when it's cheapest. And again between 10 and 17. 

I think I might release it tomorrow. Then back working on the energy sensors. 

And then next step is to start looking at calculating the amount you need to charge based on sensors and forcast etc.

Extati, 2024-11-11:
Wow, really cool - Really looking forward to the release update then. And also to the future ones! Great work!

woopstar, 2024-11-11:
> Wow, really cool - Really looking forward to the release update then. And also to the future ones! Great work!

v0.2.5 has just been released. Try it out

Extati, 2024-11-11:
Ah sorry, that make sense. I have updated the sensor. Will let you know if any problems occur.
Again, big thanks for your hard work. Appreciate it a lot.

Extati, 2024-11-12:
Wow cool - So new update later again?

Btw - Battery charged like a pro on the cheapest time last night!
Great work!!

Looking forward to future updates.

woopstar, 2024-11-12:
> Wow cool - So new update later again?
> 
> Btw - Battery charged like a pro on the cheapest time last night! Great work!!
> 
> Looking forward to future updates.

Good to hear! Keep me posted if everything works out as expected.

I am testing PR #22  currently that should solve this issue.

Extati, 2024-11-12:
Sure, will do that.

Waiting for the update then. - Big tanks

Extati, 2024-11-12:
Hmm, after I updated the program it won't work anymore. Keep initilaizing and looks like it fills out my HDD from the logs? Strange..

woopstar, 2024-11-12:
There is currently issues with Energi Data Service that causes HSEM to not work.

closed 2024-11-13 by woopstar