woopstar/hsem #24: Home Assistant Energy Sensors

woopstar, 2024-11-14:
Report bugs and/or feedback in relation to the energy sensors in this topic.

woopstar, 2024-11-15:
#23 should fix it. I will merge and release it.

Extati, 2024-11-15:
Perfect - Thanks

Extati, 2024-11-15:
Another request. Right now I dont have an EV charger sensor, our any export sensor. It looks like when it cant find the sensor, it keep filling the log about a missing sensor every second or so, so it build up the .db way to fast. Is there a fix for that?

woopstar, 2024-11-15:
> Another request. Right now I dont have an EV charger sensor, our any export sensor. It looks like when it cant find the sensor, it keep filling the log about a missing sensor every second or so, so it build up the .db way to fast. Is there a fix for that?

Can you create a new issue and put some of the logs into it, so I can see what output it is and surpress it.

Extati, 2024-11-15:
> Can you create a new issue and put some of the logs into it, so I can see what output it is and surpress it.

Sure - Will do it later

Extati, 2024-11-15:
> #23 should fix it. I will merge and release it.

Just updated, restartet and configed again, still shows working sensor as Unknown / Ukendt

woopstar, 2024-11-15:
> > #23 should fix it. I will merge and release it.
> 
> Just updated, restartet and configed again, still shows working sensor as Unknown / Ukendt

Can you provide me with logs and a screenshot of the attributes on the working mode sensor:

[image]

Extati, 2024-11-15:
What logs do you wanna see? The working sensor is empty.
Hope you can see something from these:
[image]
[image]

Extati, 2024-11-15:
In System logs i also get this error now after the update.

[image]
[image]

woopstar, 2024-11-15:
> What logs do you wanna see? The working sensor is empty. FYI: I use the standard names from Huawei solar intergration. Also my net consumption is: Unknown / Ukendt ?
> 
> Hope you can see something from this:

It did. Net consumption was not calculated because you don't have a charger. If no net consumption the sensor does not work. Can you update to 1.0.2 and see if it fixes the issue.

woopstar, 2024-11-16:
> > It did. Net consumption was not calculated because you don't have a charger. If no net consumption the sensor does not work. Can you update to 1.0.2 and see if it fixes the issue.
> 
> I have an EV Charger, but it is a Clever, so can't get it in, before I get my Shelly 3EM. Now I can't get pass the EV charger setup.
> 
> <img alt="Skærmbillede 2024-11-15 kl 21 00 58" width="351" src="https://private-user-images.githubusercontent.com/149823021/386743342-79730819-6222-466d-ba10-22052de48539.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MzE3NTk5MzksIm5iZiI6MTczMTc1OTYzOSwicGF0aCI6Ii8xNDk4MjMwMjEvMzg2NzQzMzQyLTc5NzMwODE5LTYyMjItNDY2ZC1iYTEwLTIyMDUyZGU0ODUzOS5wbmc_WC1BbXotQWxnb3JpdGhtPUFXUzQtSE1BQy1TSEEyNTYmWC1BbXotQ3JlZGVudGlhbD1BS0lBVkNPRFlMU0E1M1BRSzRaQSUyRjIwMjQxMTE2JTJGdXMtZWFzdC0xJTJGczMlMkZhd3M0X3JlcXVlc3QmWC1BbXotRGF0ZT0yMDI0MTExNlQxMjIwMzlaJlgtQW16LUV4cGlyZXM9MzAwJlgtQW16LVNpZ25hdHVyZT0zZDYxOTk3NTNmMGRlNWNkM2M5ODlkOWU3ZDgzZTNkYWIwMjhmNjBhOGM4MzA4MTFmODdhOWU4ZGJkNjg3ZWQxJlgtQW16LVNpZ25lZEhlYWRlcnM9aG9zdCJ9.aTozZ80oc1gUO3MKHbnz67gDEUCUfKeaGHVhoXRH_Ao">

can you try the latest release 1.0.3. It should be fixed now.

Extati, 2024-11-16:
> can you try the latest release 1.0.3. It should be fixed now.

Now I got past it, and looks correct whit the working sensor now.

Only thing now, is I'm not sure I use the right sensor for "Huawei solar inverter active power control state entity" We don't have any sensor named that whit the standard config from Huawei solar integration. We have "sensor.inverter_efficiency" - I assume that is the correct one?

Thanks again for a very quick fix.

woopstar, 2024-11-16:
> Now I got past it, and looks correct whit the working sensor now.
>
> Only thing now, is I'm not sure I use the right sensor for "Huawei solar inverter active power control state entity" We don't have [...]

The sensor `sensor.inverter_active_power_control` is disabled by default in the Huawei integration. You need to to the the Huawei integration and click the entities that are hidden for the inverter. Then enable the entity by clicking it.

Extati, 2024-11-16:
> The sensor `sensor.inverter_active_power_control` is disabled by default in the Huawei integration. You need to to the the Huawei integration and click the entities that are hidden for the inverter. Then enable the entity by clicking it.

Ah okay, I saw it yesterday, but didn't think that it was that. After your picture yours says limited to 100%, where mine says unlimited. I have now activated it, and make the hole setup again.

Again, thank you so much for the effort you put in this.
Also appreciate the quick replies.

Extati, 2024-11-26:
Not sure if you want me to open a issue or not, but we don't have sensor.batteries_rated_capacity under inverter or batteries that can be enabled. Only rated_capacity is under the PEES intergration, but that is from a input and not as a sensor.

Can you see were you have sensor.batteries_rated_capacity from evt.?

woopstar, 2024-11-26:
> Not sure if you want me to open a issue or not, but it looks like we don't have sensor.batteries_rated_capacity under inverter or batteries that can be enabled. Only rated_capacity is under the PEES intergration, but that is from a input and not as a sensor.
> 
> Can you see were you have sensor.batteries_rated_capacity from evt.?

I have it under Diagnostics for Batteries:

[image]

woopstar, 2024-11-26:
I'm running 1.5.0a1 of Huawei integration

Extati, 2024-11-26:
> I'm running 1.5.0a1 of Huawei integration

Ah, must be the reason then. On Public release 1.4.1 we don't have that option.

woopstar, 2024-11-26:
> > I'm running 1.5.0a1 of Huawei integration
> 
> Ah, must be the reason then. On Public release 1.4.1 we don't have that option. Maybe you can add, so both old, and new method for SOC can be used?

Let me check when 1.5.0 will be out. I think it's soon. Rather use the option. 

Until then, we could create an input helper type Number, and set it to 0 to x.000 Watts. So if you have 10kWh battery, set it to 10.000

Would that work?

Extati, 2024-11-26:
Yes, that should work in theory. 
Else you chould make it so it also could pull the data from the "Huawei_solar_pees" intergration, it uses the sensor called:

input_number.battery_rated_capacity

Don't know what makes the most sense.

woopstar, 2024-11-26:
> Yes, that should work in theory. Else you chould make it so it also could pull the data from the "Huawei_solar_pees" intergration, it uses the sensor called:
> 
> input_number.battery_rated_capacity
> 
> Don't know what makes the most sense to use.

v1.2.2 should fix this and allow input_number for the sensor.

Extati, 2024-11-26:
> v1.2.2 should fix this and allow input_number for the sensor.

Great work as always. Will test later and let you know if I have any errors.

Extati, 2024-11-26:
> Great work as always. Will test later and let you know if I have any errors.

After new version, I again can't get passed the EV charger page, I only have a sensor for an EV status sensor, and not for the power sensor. When I try to setup again, I get:

Entity is neither a valid entity ID nor a valid UUID

woopstar, 2024-11-26:
> In v1.2.2, I again can't get passed the EV charger setup-page, I only have a sensor for the EV status sensor, and not for the power sensor. When I try to set it up again, I get:
> 
> Entity is neither a valid entity ID nor a valid UUID
> 
> The error also comes if you don't enter anything.

Can you try v1.2.3 now

Extati, 2024-11-26:
> Can you try v1.2.3 now

Yes, now I got pass the setup. The only noticeable thing under the working mode is this:

Batteries conversion loss: Ukendt
Batteries remaining charge: Ukendt

That is not right, is it?

Extati, 2024-11-26:
> Can you screenshot the configuration screens and also all attributes ? Might take a minute or two for the sensor to adapt. But the rated capacity is missing.

The sensor has not updated yet, I think it's the input helper not giving the data it's needed.

Screenshots of the sensors is here:

 
[image]
[image]
[image]

Extati, 2024-11-26:
> Edit: Just tried to use another sensor for the EV charger to get pass it, and set it up again, seams that you can't change the sensor to and input. So I think the error is there somewere.

Back on v1.1.1 again, and all sensors and work mode is fine again. Will wait until further updates.

woopstar, 2024-11-27:
> Back on v1.1.1 again, and all sensors and work mode is fine again. Will wait until further updates.

I found the issue and will soon release 1.2.4, which fixes it. Thank you for testing everything!

Extati, 2024-11-27:
> I found the issue and will soon release 1.2.4, which fixes it. Thank you for testing everything!

Great work. Sure, it's a pleasure to test, and see the work you are implementing, to make it better all the time.

Extati, 2024-11-27:
> I found the issue and will soon release 1.2.4, which fixes it. Thank you for testing everything!

Updated to 1.2.4. and now it look more right. The only thing that confuses me is the Batteries conversion loss, is still shows as: Ukendt

Also the "Huawei solar batteries tou charging and discharging periods periods" shows:  00:00-00:01/1234567/+ and not the time like in the old versions. Don't know if this is intended or not?

See below sensor settings now in .1.2.4:

[image]
[image]
[image]

Extati, 2024-11-27:
> The TOU mode is OK. It should change later on.

Indeed it did - so far all seams good - I would say 1.2.5 is stable and working perfect so far.

Extati, 2024-12-11:
Just for everyone to know, version 2.0 is super great, stabe and charging/discharging is so far spot on.
Great work as always.

Extati, 2024-12-18:
Update 2.1.0 have a fault when you update from a working intergratin you get an error saying that you are missing inputs, when you try to config, you again can't get pass the EV charger page..

woopstar, 2024-12-18:
> Update 2.1.0 have a fault when you update from a working intergratin you get an error saying that you are missing inputs, when you try to config, you again can't get pass the EV charger page..

2.1.1 should fix the issue. Just published it.

Extati, 2024-12-18:
> 2.1.1 should fix the issue. Just published it.

Indeed it did - Thanks again for the quick fix as always.

Extati, 2025-02-21:
After update 3.0 all my sensors show just 0.0 kwh. The only one that shows something hourly is utility_meter hourly. It will not buy any power from the grid, even if all inputs are filled correctly. Anyone else experiencing the same error? I have deleted the integration and set it up again, but the same error and sensor logic occurs again.

Also the sensor for the active hour seams to be very High. 

[image]
[image]

Extati, 2025-02-21:
Update: Back on version 2.2.0 and everything is fine again - All sensors show correct data now.

Where is the data keep's from the sensor for the diffrent days? I tried to purge the hole HA database, but it somehow still know some of the sensor historic on both version 3.0 and on the old 2.2.0 - Just wanted to test if it helps to clear some of the sensor, but on version 3.0 it still shows 0 on most of the sensors as above picture, trust that HSEM display OK.

For information I'm on version 1.5.4 on the Huawei Solar integration, but that shouldn't effect anything despite version 2.2.0 works. Can't seam to find why version 3.0 don't work on my system.

heinoskov, 2025-02-21:
In developer tools --> Statistics - you can search for HSEM and select those you want to delete..

woopstar, 2025-02-21:
> After update 3.0 all my sensors show just 0.0 kwh. The only one that shows something hourly is utility_meter hourly. It will not buy any power from the grid, even if all inputs are filled correctly. [...]

How long did you run for ? V3.0.0 introduces new energy sensors as the ones in 2.2.0 is wrong. This also means when you upgrade the sensors will reset and it will take up to 14 days before they find themselves again.

woopstar, 2025-02-21:
> I ran the new version since release? Just funny why they are unavailable or is it just me? I will wipe the hole integration now, delete all old sensors and try version 3.0.0 again. Just saw the we didn't purchased any power doing night and day, and I would assume it already would have done that?

The power sensors are unavailable when they are not measuring. 

If they are reporting 0.0 as state, it will be part of the calculations giving wrong energy calculations. So they are intentionally unavailable to make sure counts / measurements stops in the particular hour.

You should look into the Energy Avg sensors. They should start to have value as time passes.

Extati, 2025-02-21:
> The power sensors are unavailable when they are not measuring.
>
> If they are reporting 0.0 as state, it will be part of the calculations giving wrong energy calculations. So they are intentionally [...]

Yes, but after 3 days the day 1 sensor and day 3 still showed 0 - I can also see in the log, that it is schedule 1 that has a little bit of data, and not schedule 2 - I don't know where the error is. I will try to run it a day or two more, but I don't think it will help.

This is the log so far..

[hsem.log](https://github.com/user-attachments/files/18915552/hsem.log)

Extati, 2025-02-22:
Update: The log is now going haywire, is already over 10 MB, after a couple of hours, it started to log almost every second.
As you can see, it calculate not to charge the battery, despite there is a small demand and the battery is at 0.

Also all day 1 sensores still shows 0.0 - Version 2.2.0 started filling the sensors a bid after each hours.
Do you need med to do or post anything else, please just let me know.

[hsem.log](https://github.com/user-attachments/files/18920842/hsem.log)

woopstar, 2025-02-22:
> Update: The log is now going haywire, is already over 10 MB, after a couple of hours, it started to log almost every second.
> As you can see, it calculate not to charge the battery, despite there is a small demand and the battery is at 0.
> 
> Also all day 1 sensores still shows 0.0 - Version 2.2.0 started filling the sensors a bid after each hours.
> Do you need me to do or post anything else, please just let me know.
> 
> Edit: It has now made another log file. The first one has over 90.000 lines of debug code already.
> 
> First log:
> [hsem.log](https://github.com/user-attachments/files/18921017/hsem.log)
> New log:
> [hsem.log](https://github.com/user-attachments/files/18921023/hsem.log)

Great. I'll have a look. 

Can you also send me the content of home-assistant.log, and also from developer tools search for HSEM and send me a screenshot of the HSEM sensors with its attributes.

Extati, 2025-02-22:
Sure.

Log is here:

[home-assistant.log](https://github.com/user-attachments/files/18922587/home-assistant.log)

Screenshots of some of the sensors. Let me know if you need specific ones. As you can see, the utility_meter shows something, but the energy_avg sensor all days are 0.0

[image]
[image]
[image]
[image]

Again, really appreciate the support you put into this.

woopstar, 2025-02-23:
@Extati 

I think I've found the issue. From the average sensor attributes, you see the tracked entity. This utility meter should exist. Can you verify that?

But I also see that Measurements are "Ukendt," and this is a bug.

Can you manually try to change this in your HA installation, as I think you can do so:

Open the file in the `/config` folder:

custom_components/hsem/custom_sensors/avg_sensor.py

Go to the line: 176, that contains:

```python
if utility_meter_value is not None and self._measurements is not None:
    self._measurements[current_date.isoformat()] = round(
        float(utility_meter_value), 2
    )
```

And change it to:

```python
if self._measurements is None:
    self._measurements = {}

if utility_meter_value is not None:
    self._measurements[current_date.isoformat()] = round(
        float(utility_meter_value), 2
    )
```

Then restart HA. Then you should be able to see the measurements variable contains data, as you can see from this screenshot:

[image]

Extati, 2025-02-23:
Yes I see the tracked entity from the utility_meter on each hour.

I changed the code you send me and now all the "day" sensors show data - Yay!

After the restart the battery also is charging now - And make sense as I right now charge my EV and sun is really not there to fill out the battery before 17 - So I assume it works!?

What about the stop of battery usage happening at 23 and not on 23:59 as I have specific in the schedule 2?

A BIG thanks so far for finding the error.

woopstar, 2025-02-23:
> Yes I see the tracked entity from the utility_meter on each hour.
>
> I changed the code you send me and now all the "day" sensors show data - Yay!
>
> After the restart the battery also is charging now - [...]

Awesome. I'll prepare a fix for release. 

In regards to the schedules, they actually don't work with minutes yet 😆 I'll start looking into fixing that as I have the same issue. My schedule is also 17.00-23.59

I'm also considering to make a fix so you can schedule 17.00-00.00 for instance. Or 17.00-03.00

Extati, 2025-02-23:
> Awesome. I'll prepare a fix for release.
>
> In regards to the schedules, they actually don't work with minutes yet 😆 I'll start looking into fixing that as I have the same issue. My schedule is also [...]

Haha, okay - It make sense why it stops at 23:00 then 😄 

It would be really great it we could set it from 17:00 - 0:00 that would cover most of the "high" prices.

Great work as always, and appreciate the quick fix and support you put into this.

ermitovski, 2025-09-07:
Hi! Following this https://github.com/woopstar/hsem/pull/162 regarding winter/spring months, would it be possible to make it configurable per user? 
In my case, I could extend the "summer" period to include October and April.

woopstar, 2025-09-07:
> Hi! Following this [#162](https://github.com/woopstar/hsem/pull/162) regarding winter/spring months, would it be possible to make it configurable per user? In my case, I could extend the "summer" period to include October and April.

Yes. Let me change that so it can be configured. Will see how it fits best to select the months in the config flow.

Extati, 2025-09-14:
> Can you test the latest beta release :)

v.4.0-Beta1 dont seam to work.

Logger: homeassistant.config_entries
Kilde: config_entries.py:751
Første forekomst: 11.24.14 (1 forekomst)
Senest logget: 11.24.14

Error setting up entry Huawei Solar Energy Management for hsem
Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/config_entries.py", line 751, in __async_setup_with_context
    result = await component.async_setup_entry(hass, self)
             ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/config/custom_components/hsem/__init__.py", line 64, in async_setup_entry
    await hass.config_entries.async_forward_entry_setups(entry, PLATFORMS)
  File "/usr/src/homeassistant/homeassistant/config_entries.py", line 2571, in async_forward_entry_setups
    await integration.async_get_platforms(platforms)
  File "/usr/src/homeassistant/homeassistant/loader.py", line 1173, in async_get_platforms
    import_future.result()
    ~~~~~~~~~~~~~~~~~~~~^^
  File "/usr/src/homeassistant/homeassistant/loader.py", line 1173, in async_get_platforms
    import_future.result()
    ~~~~~~~~~~~~~~~~~~~~^^
  File "/usr/src/homeassistant/homeassistant/loader.py", line 1173, in async_get_platforms
    import_future.result()
    ~~~~~~~~~~~~~~~~~~~~^^
  [Previous line repeated 1 more time]
  File "/usr/src/homeassistant/homeassistant/loader.py", line 1161, in async_get_platforms
    platforms.update(self._load_platforms(platform_names))
                     ~~~~~~~~~~~~~~~~~~~~^^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/loader.py", line 1086, in _load_platforms
    platform_name: self._load_platform(platform_name)
                   ~~~~~~~~~~~~~~~~~~~^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/loader.py", line 1260, in _load_platform
    cache[full_name] = self._import_platform(platform_name)
                       ~~~~~~~~~~~~~~~~~~~~~^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/loader.py", line 1292, in _import_platform
    return importlib.import_module(f"{self.pkg_path}.{platform_name}")
           ~~~~~~~~~~~~~~~~~~~~~~~^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/util/loop.py", line 201, in protected_loop_func
    return func(*args, **kwargs)
  File "/usr/local/lib/python3.13/importlib/__init__.py", line 88, in import_module
    return _bootstrap._gcd_import(name[level:], package, level)
           ~~~~~~~~~~~~~~~~~~~~~~^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "<frozen importlib._bootstrap>", line 1387, in _gcd_import
  File "<frozen importlib._bootstrap>", line 1360, in _find_and_load
  File "<frozen importlib._bootstrap>", line 1331, in _find_and_load_unlocked
  File "<frozen importlib._bootstrap>", line 935, in _load_unlocked
  File "<frozen importlib._bootstrap_external>", line 1026, in exec_module
  File "<frozen importlib._bootstrap>", line 488, in _call_with_frames_removed
  File "/config/custom_components/hsem/sensor.py", line 9, in <module>
    from custom_components.hsem.custom_sensors.working_mode_sensor import (
        HSEMWorkingModeSensor,
    )
ImportError: cannot import name 'HSEMWorkingModeSensor' from 'custom_components.hsem.custom_sensors.working_mode_sensor' (/config/custom_components/hsem/custom_sensors/working_mode_sensor.py)

woopstar, 2025-09-15:
> v.4.0.0-Beta1 dont seam to work.

Yes. There was a bug. Released beta2 of 4.0.0 that works now.

Extati, 2025-09-15:
> Yes. There was a bug. Released new beta of 4.0.0 that works now.

Great - It also looks like latest beta work whit fully fresh install. Will test it out!

woopstar, 2025-09-17:
> I'm really impressed by this 4.0.0 version. The hole integration feels way more "alive" and you can see how the solarpanels change workflow doing the hole day, instead of planned daywork before.
> 
> The only "complain" is after a restart of HA, the first 15 minuts it looks like the sensors don't work, as the working sensor first comes alive after 15 minutes again. Else this version is a game-changer so far.

Did you check the attributes while the 15 minutes and complaint. It should now tell you which sensors it’s missing / waiting for.

Extati, 2025-10-16:
> Good catch. I released v4.0.0-beta8, which should fix the above error. Can you test?

Yes, It looks like that did the trick.

Thanks!

woopstar, 2025-10-16:
> Yes, It looks like that did the trick.
> 
> Thanks!

Great, so it resolved all of your problems?

Extati, 2025-10-16:
> Great, so it resolved all of your problems?

Yes, it fixed the above problem whit the working sensor not working after restart.

But I think I found another "bug". When the batteries are in "waiting mode" (in the night) when my EV charger starts to charge the car. The allowed battery discharge (allowed to cover the houseload) the batteries starting to discharge due to the working sensor change to EV_smart_charging. Is there some way to change that logic, when the batteries came from "waiting_mode"? Basically discharge of battery while EV_smart_charging is active should perhaps only work in the summer months maybe? Does that make sense?

woopstar, 2025-10-17:
> Yes, it fixed the above problem whit the working sensor not working after restart.
>
> But I think I found another "bug". When the batteries are in "waiting mode" (in the night) when my EV charger [...]

Ah. So batteries wait mode does not take precedence over the ev smart charging mode.

Extati, 2025-10-17:
> Ah. So batteries wait mode does not take precedence over the ev smart charging mode.

Nope - Lets take example from last night.

Plan from 17-23.30 is active and batteries where in "discharge_mode"
When we hit 23.30 we had 25% battery left and that was enough to cover the morning plan from 6-11 as we would start to produce power around 8.15 ish. So the batteries would have covered that time and no need for grid charge in the night.
But around 1 o clock, the EV charger started to charge and then the batteries went from Wait_mode to EV_smart_charging and then we have implemented that the batteries are allowed to cover houseload (350watt in my case) So the batteries died around 5 o'clock clock. Why they don't begging to charge from grid to cover the morning plan around 6 I don't know either. In fact I havet seen "charge_from_grid" in the version 4.0 I believe.

So the solution for the battery discharge when EV charge should only be in the summer months or if the batteries came from "discharge_mode" perhaps?

woopstar, 2025-10-20:
> But maybe a button could make sense for an automation, if it’s not too much work for you.

v4.0.0-beta9 released with the switch :)

[5 more replies]