woopstar/hsem #43: New sensor type for EV charging?

Extati, 2024-11-20:
Would it make sense to allow users to select a Boleen sensor instead of the sensor.xxx during the setup of EV Chargers?

This would make it easier to create a virtual sensor, allowing users to, for example, "turn on" their electric vehicle charger, and the integration would automatically ensure that the battery doesn't get drained.

It might also be possible to implement this in a simpler way, but many users today have Clever chargers, where unfortunately, it's not possible to access the charger status.

woopstar, 2024-11-21:
I'm not entirely sure what you mean. Isn't this what the EV Charger Status Sensor input sensor does?

What you ask for is that you can only add a sensor right now. What you want to do is also to be able to add an input_boolean from a helper/template

Extati, 2024-11-22:
> What you ask for is that you can only add a sensor right now. What you want to do is also to be able to add an input_boolean from a helper/template

Yes, correct - That would make it much easier for us whit an charger who can't easily be implemented in HA.

Sorry if I wasent clear.

PR #47 (2024-11-22): fix: allow switch and input booleans for ev charger status
Fixes #43
closed 2024-11-22 by woopstar