woopstar/hsem #69: Update priority of when to charge battery

woopstar, 2024-11-29:
We should consider updating the time to charge the battery on more inputs. Currently, it only looks for the cheapest hours.

We should update it to sort and look at net consumption and negative import prices.

Negative import prices should be prioritized because we get money to charge our batteries. Secondly, we have a surplus from our PV if we have negative net consumption.

Extati, 2024-11-29:
Charging batteries when negative import price is next level. We still just need to look at the amount needed for "high price" times - like from 17-22 and 06-09 etc.

woopstar, 2024-11-29:
> Charging batteries when negative import price is next level. We still just need to look at the amount needed for "high price" times - like from 17-22 and 06-09 etc.

It will currently only look for negative import price in the ranges defined.

In regards to 17-21 and 06-09 etc, my plan was to look into:

The price it costs to charge the battery to support that time, and that we need to charge at least 4-5 kWh before we actually force charge the battery.

So overall, my goal is to look into "what conditions" should be met before we actually force charge the battery, and next, if we need to force charge it, then when.

Extati, 2024-11-29:
> Oh. Btw. My beta 2.0.0 that is in the PR has already implemented the above 3 sorted priorities

Nice, that would be a great benefit for the hole system. Can we force discarge to gird on negative prices also, if we can calculate that we can get the amount needed again from the sun before peek hours, or if we can calculate that it makes more money to sell/discarge now, and buy cheaper later? or does the system almost do that now? Haven't studied the hole codes yet 👍

PR #68 (2024-12-06): fix: v2.0.0
Fixes #40 
Fixes #41 
Fixes #63 
Fixes #64 
Fixes #65
Fixes #69
Fixes #70 
Fixed #72 
Fixes #73 

The battery charge time in day and night will now be prioritized by the following:

**Sort hours by priority:**
1. Negative import price
2. Surplus (net_consumption < 0)
3. Lowest import price
closed 2024-12-06 by woopstar