woopstar/hsem #84: Rated Capacity of battery pack

HolgerMiara, 2025-01-04:
Hi Andreas, 

You write in the readme that users of your HSEM shall enable the `sensor.batteries_rated_capacity`. As requested, I have installed **version 1.5.0a1** of the [Huawei Solar integration](https://github.com/wlcrs/huawei_solar), but the integration doesn't offer any sensor which shows the capacity of the connected battery pack. I have now created a helper with the same name and set its state to 5 (kWh). 

Any explanation why you appear to have it and I not?

Cheers, 
Holger.

woopstar, 2025-01-05:
> but the integration doesn't offer any sensor which shows the capacity of the connected battery pack.

This is because the entity is disabled by default. Using an input helper also works, but the sensor exists in 1.5.0a1.

It's found in the Diagnostic section:

[image]

HolgerMiara, 2025-01-05:
Thanks a million!
No idea why I didn't find it yesterday..?
Have a nice Sunday, 
H.

closed 2025-01-05 by HolgerMiara