woopstar/hsem #105: Charge from grid during night despite sun in morning

jonaslevring, 2025-02-18:
First.. the hsem is really working great :-)

But there seems to be an issue around transition from vinter to spring. Days where where there are lots of sun during the daytime, hsem still charges the battery from grid to 100% during night.

I would expect it to just charge enough from grid during night to survive till the the sun produces enough sun to charge the battery.
Yesterday only 0,8kW was free on battery when the panels started to produce enough..  forecast said 27kW, and i exported 35kW during the day.

Hope it can be fixed

Extati, 2025-02-21:
I'm experiencing the same thing. It seems like there might be some errors in the way it calculates the required energy. See my post in the feedback section. I'm not sure if your sensors show the same "0" as mine do..

woopstar, 2025-02-21:
> I'm experiencing the same thing. It seems like there might be some errors in the way it calculates the required energy. See my post in the feedback section. I'm not sure if your sensors show the same "0" as mine do..
> 

Can you update to 3.0.0 again and let it run for at least 24-48 hours. Then send me the HSEM log file after you have enabled Verbose Logging on the device. It will be the HSEM.log file in /config

Extati, 2025-02-21:
> Can you update to 3.0.0 again and let it run for at least 24-48 hours. Then send me the HSEM log file after you have enabled Verbose Logging on the device. It will be the HSEM.log file in /config

Yes, I'm up and running again - the log is filling, so will poste in a day or two - Thanks for taking the time to look into it. Much appreciated.

jonaslevring, 2025-02-21:
has been running 3.0 for 2 days now.. working better the first days, but has stopped now.
Has enabled verbose logging, but /config/hsem.log is empty

woopstar, 2025-02-23:
> has been running 3.0 for 2 days now.. working better the first days, but has stopped now. Has enabled verbose logging, but /config/hsem.log is empty

v3.0.1 should fix the issue. Can you try to update.

jonaslevring, 2025-02-24:
Already looks a lot better this morning.. will let it run for a few days and then close issue.. good job 👍

closed 2025-02-26 by jonaslevring