woopstar/hsem #112: Prefer export while price is high

jonaslevring, 2025-03-02:
Idea: If there is a good forecast that would easily fill battery, it would be great if battery charge could be postponed.. Simply to utilizy best possible export prices.
It should open up for battery charge around the worst export prices to have maximum effect.

One thing that could disturb this proces is if car charge is needed. But maybe that could be managed through an entity if it should optimize for "battery-then-export" or "export-then-battery"

jbruntt, 2025-03-10:
I agree. This would be nice. I do not think I have the skills to code that kind of thing, but a place that one could seek inspiration is in the [EV Smart Charging Integration](https://github.com/jonasbkarlsson/ev_smart_charging/) that calculates when the optimal time is to charge an electric car.

woopstar, 2025-03-20:
This is a good idea. I will try to see how we can utilize this.

woopstar, 2025-07-16:
Currently working on this one. Waiting on tests to complete, as the weather is presently not in favour of this case.

closed 2025-08-11 by woopstar

woopstar, 2025-08-12:
@jonaslevring @Extati This should work now in v3.3.0

Extati, 2025-08-12:
> [@jonaslevring](https://github.com/jonaslevring) [@Extati](https://github.com/Extati) This should work now in v3.3.0

So it will sell at a high export price, when knowing that the battery can easily be recharged later? That could be really cool.

jbruntt, 2025-08-12:
> So it will sell at a high export price, when knowing that the battery can easily be recharged later? That could be really cool - Will test this and keep an eye on the export.

@Extati I believe that this change does something different.
If today is a high production day (according to forecast), it plans when it is the best to charge the battery. 
"Best" = when the sale-price is the lowest.
The charging might be postponed from early hours to mid-day.

woopstar, 2025-08-12:
> it would sell sometimes even before the batteries were full

It will

> what are your settings for export price? I just use the "standard" but not sure if there is a better setting.

I use the standard too.

PR #140 (2025-08-11): feat: Add configuration options for EV charger discharge power
Introduce new configuration options for maximum discharge power and force maximum discharge power for the EV charger. Update related settings and documentation for improved sensor handling.

Fixed #144 
Fixes #139
Fixes #112
Fixes #107
Fixes #85
Fixes #74