woopstar/hsem #113: Export on negative prices

jonaslevring, 2025-03-08:
Theres is an issue with export.. It exports despite prices are negative.

Maybe a custom tresholds for when to cutoff export 

[image]

woopstar, 2025-03-20:
I see. I will implement a threshold you can set in the configuration so the export price MUST exceed some limit. As of now, it would start the export if it's -0,01

closed 2025-05-09 by woopstar

woopstar, 2025-05-12:
This is fixed in [v3.1.1](https://github.com/woopstar/hsem/releases/tag/v3.1.1)

In the configuration flow, you can now set the minimum required export price before you stop the solar system. It currently defaults to 0.15 DKK/kWh.

PR #124 (2025-05-09): feat: add minimum export price configuration for energy data service
Fixes #113 

This pull request introduces a new configuration parameter, `hsem_energi_data_service_export_min_price`, to enhance the energy data service functionality by allowing users to define a minimum export price. The changes span across constants, sensor logic, UI schema, and translations to fully integrate this feature.

### New Feature: Minimum Export Price Configuration

* **Constants Update**:
  - Added `hsem_energi_data_service_export_min_price` with a default value of `-0.15` in `custom_components/hsem/const.py`.
 [... cut]