woopstar/hsem #114: Charge controls

ermitovski, 2025-03-08:
Hi,
I'm using version 3.0.1 and I can't see the battery charging controls https://github.com/woopstar/hsem/issues/72 implemented in this PR
Is it possible to disable the daytime charge with this version?

These are the current controls I can see
[image]

Thanks!

woopstar, 2025-03-20:
Hi.

The charge controls for night and day was ditched in favor of the battery schedules instead. The same functionality can be acquired by setting two schedules - one for night and one for day.

It's a bit hard to see, and I've made the mistake multiple times, but make sure start time and end time is set correctly. I have set the start time in the end time field multiple times.

ermitovski, 2025-03-23:
Thanks for the explanation! I'll play a bit with the config, I think I got confused with the new version and some old references.

closed 2025-05-10 by woopstar