woopstar/hsem #119: Dry-run mode

ermitovski, 2025-03-23:
Hi,
As a new user, enabling the integration is a bit intimidating since it immediately makes changes to the batteries configuration.

I noticed a comment in another issue mentioning that the integration needs to run for 14 days before it has sufficient sensor data.

Would it make sense to implement a "dry-run mode," especially for the first 14 days, that would show the proposed configuration changes for the batteries without actually applying them?

woopstar, 2025-03-31:
There is already a "read-only" switch. Would that work as needed?

ermitovski, 2025-03-31:
Exactly that! I missed that option the first time I configured the integration. Thanks!