woopstar/hsem #127: Bug: No surplus when export price is negative

woopstar, 2025-05-12:
Encountered this weekend that the power control was activated to limit solar production due to negative export price. This had a side draw, that there was no surplus available to charge the EV from surplus power, since the inverter was limited.

Need to fix this, so when export price is negative and the battery is charged, the power control is not activated if we are able to charge the car also.

Temp hack to fix:

Update to pre-release v3.1.1 that includes option to set minimum export price before activation of power control.
Set min export price to something high to prevent the power control.

jonaslevring, 2025-10-05:
seems like this has not been solved yet.. Now you got the sensor for EV charging working, it could be possible to just to allow export when EV charger is on..

My EV charger sensor is just a switch turning on if power is > 100w (larger than standby power). So you could take the power sensor for charger directly and make similar logic to ensure export.. but there will naturally always be issues where you can export because your charger cannot adapt to low amps..

But overall there are really some great opportunities to make some nice integrations with the EV Smart Charging integration.

jonaslevring, 2025-10-05:
i understand your thoughts behind.. but remember there are multiple older EV cars that are recommended to charge to 80%.. It would then be the car stopping the charging at eg 80%, and you would still have "Car connected" = true, but not charging.. which would result in wrongly export

woopstar, 2025-10-05:
> i understand your thoughts behind.. but remember there are multiple older EV cars that are recommended to charge to 80%.. It would then be the car stopping the charging at eg 80%, and you would still have "Car connected" = true, but not charging.. which would result in wrongly export

So we would need the target SoC for the car also. But why would you not charge above 80% of its free even? Or maybe add a switch to allow charge past the target SoC?

woopstar, 2025-10-16:
@jonaslevring Did it work out? Waiting for feedback on this one before we release 4.0