woopstar/hsem #130: Idea - HSEM brakes  down on "missing inputs"

Extati, 2025-05-16:
This isn't exactly a bug with the integration, but more of a suggestion or a request for a safety mode.

As many are probably aware, Energidataservice is currently "down" due to changes in how they deliver data. Mtrab is unable to resolve the issue quickly, so we need to wait for a proper solution.

Currently, the entire HSEM integration is non-functional due to missing input data — specifically, the price.

Is there a way we could make the integration work without having access to the actual price data? Perhaps by using a fixed price or another fallback solution?

woopstar, 2025-06-25:
@Extati 

Would it work to have a drop down select on the device to "Force Working Mode" with the options:

"Auto"
"Force Export"
"Batteries Wait Mode"
"Batteries Charge Grid"
"Batteries Charge Solar"
"Batteries Discharge Mode"

Where as Auto is default and would lead to the above issue with missing input entities, but you can override the state via the drop down?

What options would you like to be available?

Extati, 2025-06-25:
Also an idea – how about an option to set a maximum battery discharge rate to cover the house load during the night while the EV is charging?

Many people use EV chargers that reimburse the charging cost, but since charging typically happens at night, the battery doesn’t assist with the household load. If we could define a custom battery discharge limit specifically for when EV charging is active, it could help cover the basic night-time household consumption

Does it make sense? :-)

woopstar, 2025-06-26:
> Also an idea – how about an option to set a maximum battery discharge rate to cover the house load during the night while the EV is charging?
>
> Many people use EV chargers that reimburse the charging [...]

First, charging the battery, then letting it help charge the battery in the EV is not worth it, especially due to the conversion loss. So, from an economic point of view, it's not feasible to let the battery help charge the EV.

There might be "some" option to allow it to discharge the same amount as the house consumes, to cover the house consumption, while the extra import will only be for the EV charger.

Extati, 2025-06-26:
> There might be "some" option to allow it to discharge the same amount as the house consumes, to cover the house consumption, while the extra import will only be for the EV charger.

That would be epic if this was possible.

woopstar, 2025-06-26:
> It is a principle i've presonally just done in node-red, but it works fine, and it would be great to have it integrated.. Please be aware about update times.. Once XX second/minute, because a constant change overloads the Huawei dongle.. Every minute works fine for me. Typically not an issue during charge at night at house power consumption typically is stable, but during day with people in house (fluctuations in consumption), EV charging and regulating EV amp up/down to consumption.. then be becomes a bit more tricky.
> 
> But overall a good idea, just make dongle update times adjustable and a custom house sensor to update from.

The dongle will only get updates as per the working mode update interval configured. So every 5 min etc.

Extati, 2025-06-27:
> Should work in pre-release v3.2.0-beta1. Will do some testing.

Cool - Will wait for a release then.

woopstar, 2025-06-27:
> Cool - Will wait for a release then - Thanks!

You can go to HACS, select HSEM, Update Information and select Redownload, then in the Release dropdown, you can select the pre-release and test it.

woopstar, 2025-06-28:
Found a few issues, but did try to charge my EV during the night. And Batteries Discharge Power is now being set to net consumption:

[image]

This is fixed in v3.2.0-beta2

Extati, 2025-06-29:
> Found a few issues, but did try to charge my EV during the night. And Batteries Discharge Power is now being set to net consumption:

Also just updated to v3.2.0-beta2 - so far jo good. But how does it work whit max discharge power? I don't have a sensor that measure the power for the EV, only a sensor that says it EV is charging or not defined by a sensor that measure if the power on the garage goes over a fixed limit.

Is there a way I can set that if EV charger is on, the max discharge power from battery should be 350 watt, as this would cover the basis house consumption doing the night?

Extati, 2025-07-05:
So far everything is working.