woopstar/hsem #142: Unexpected Battery Discharge During ev_smart_charging Mode

jbruntt, 2025-08-08:
Hi,
I've observed that sometimes when sensor.hsem_workingmode_sensor is in the ev_smart_charging state, the battery connected to my inverter starts discharging. I believe this behavior is not intended.

To clarify my setup:
- I have not configured the integration to allow discharge.
- **EV Charger Force Max Discharge Power** is **off**.
- **EV Charger Max Discharge Power** is set to **500**.
Let me know if any additional logs or configuration details would help in diagnosing this.

Thanks for your work on this integration!
[image]

Extati, 2025-08-09:
What version of HA are you running? I updated to newest one, and HSEM broke down, so sensors worked, and all data logging sensor was not available anymore.

Huawei also failed on the battery parameters.

If you are on the newest, to roll back and see if it helps?

Else, how many Watt/Kwh is your battery outputting?
What does the HSEM status sensor says?

If you can provide a bit more info, then we can try to help.

Extati, 2025-08-11:
> * Battery Working Mode: maximise_self_consumption

When you charge, it should be in: TimeOfUse, if I'm not mistaken? The error seems to be with your battery sensor. Is the correct entity chosen in the config?

jbruntt, 2025-08-11:
I updated HSEM to 3.3.0.  I still see the discharging of the battery while charging my EV.

I checked the battery sensor I had configured HSEM to use.
Just to make sure. We are talking about this configuration item? See image below.
[image]
The sensor I have configured HSEM to use is called `select.batteries_working_mode`. It is a sensor exposed by the Huawei integration.
I just had a look at the sensor in Developer tools->States. At this time it has State: **maximise_self_consumption** and these attribute-values: 
options: adaptive, fixed_charge_discharge, maximise_self_consumption, fully_fed_to_grid, **time_of_use_luna2000** friendly_name: Batteries Working Mode
I noticed the "time_of_use_luna2000" which is a bit odd. I believe that the battery type of my installation is called Luna2000.
Could it be that HSEM does not know this odd naming of the State?

Extati, 2025-08-12:
Time_of_use_luna2000 is the mode your battery should be in when you charge your EV.
You have a sensor for your EV Charger right? Else the intergration will not know when you charge.

If you dont have, thats the reason.

woopstar, 2025-08-12:
ev_charger_status_state is false in your output.

Can you enable the switch Extended Attributes, wait until working mode sensor has updated, and paste the attributes again. That should give us the information.

jbruntt, 2025-08-13:
I made a template for collecting information.
I do (and did) have a sensor configured for when the EV is charging.
The below information is collected more than 5 minutes after my EV started charging.
Looks to me like HSEM does see that we are charging. But the Huawei integration says that the battery is doing maximise_self_consumption.

HSEM v3.3.0 Debug Information:

=== SENSOR DATA ===
Battery Working Mode (Huawei Direct): maximise_self_consumption
Sensor: select.batteries_working_mode

HSEM Working Mode: ev_smart_charging
Sensor: sensor.hsem_workingmode_sensor

EV Charger Status State (HSEM): True
Attribute: sensor.hsem_workingmode_sensor -> ev_charger_status_state

Battery Working Mode (HSEM View): maximise_self_consumption
Attribute: sensor.hsem_workingmode_sensor -> huawei_solar_batteries_working_mode_state

EV Status (Home Assistant): charging
Sensor: sensor.carport_jens_bruntt_status

EV Power (Home Assistant): 2.20499992370605W
Sensor: sensor.carport_jens_bruntt_power

=== BATTERY OPTIONS ===
Available Battery Working Modes: ['adaptive', 'fixed_charge_discharge', 'maximise_self_consumption', 'fully_fed_to_grid', 'time_of_use_luna2000']

=== POWER STATUS ===
Battery Discharge Power: 0W
Battery Charge/Discharge Power: 193W
Solar Production: 3357.0W
Battery SOC: 63.0%

=== ALLE HSEM WORKING MODE ATTRIBUTES ===
batteries_conversion_loss: 10.0
batteries_current_capacity: 5.8
batteries_usable_capacity: 9.5
energi_data_service_export_entity: sensor.energi_data_service_sale
energi_data_service_export_min_price: -0.22
energi_data_service_export_state: 0.648
energi_data_service_import_entity: sensor.energi_data_service
energi_data_service_import_state: 2.049
energy_needs: {'0am_6am': 0.0, '6am_10am': 0.0, '10am_5pm': 0.0, '5pm_9pm': 0.0, '9pm_midnight': 0.0}
ev_charger_force_max_discharge_power: False
ev_charger_max_discharge_power_state: 500
ev_charger_power_entity: sensor.carport_jens_bruntt_power
ev_charger_power_state: 2.2
ev_charger_status_entity: sensor.carport_jens_bruntt_status
ev_charger_status_state: True
force_working_mode_entity: select.huawei_solar_energy_management_force_working_mode
force_working_mode_state: auto
hourly_calculations: {'00-01': {'avg_house_consumption': None, 'solcast_pv_estimate': 0.0, 'estimated_net_consumption': None, 'estimated_cost': 0.0, 'batteries_charged': 0.0, 'import_price': 2.06, 'export_price': 0.696, 'recommendation': 'time_passed'}, '01-02': {'avg_house_consumption': None, 'solcast_pv_estimate': 0.0, 'estimated_net_consumption': None, 'estimated_cost': 0.0, 'batteries_charged': 0.0, 'import_price': 2.006, 'export_price': 0.653, 'recommendation': 'time_passed'}, '02-03': {'avg_house_consumption': None, 'solcast_pv_estimate': 0.0, 'estimated_net_consumption': None, 'estimated_cost': 0.0, 'batteries_charged': 0.0, 'import_price': 1.977, 'export_price': 0.63, 'recommendation': 'time_passed'}, '03-04': {'avg_house_consumption': None, 'solcast_pv_estimate': 0.0, 'estimated_net_consumption': None, 'estimated_cost': 0.0, 'batteries_charged': 0.0, 'import_price': 1.957, 'export_price': 0.614, 'recommendation': 'time_passed'}, '04-05': {'avg_house_consumption': None, 'solcast_pv_estimate': 0.0, 'estimated_net_consumption': None, 'estimated_cost': 0.0, 'batteries_charged': 0.0, 'import_price': 1.979, 'export_price': 0.632, 'recommendation': 'time_passed'}, '05-06': {'avg_house_consumption': None, 'solcast_pv_estimate': 0.005, 'estimated_net_consumption': None, 'estimated_cost': 0.0, 'batteries_charged': 0.0, 'import_price': 2.037, 'export_price': 0.678, 'recommendation': 'time_passed'}, '06-07': {'avg_house_consumption': None, 'solcast_pv_estimate': 0.228, 'estimated_net_consumption': None, 'estimated_cost': 0.0, 'batteries_charged': 0.0, 'import_price': 2.232, 'export_price': 0.794, 'recommendation': 'time_passed'}, '07-08': {'avg_house_consumption': None, 'solcast_pv_estimate': 0.915, 'estimated_net_consumption': None, 'estimated_cost': 0.0, 'batteries_charged': 0.0, 'import_price': 2.219, 'export_price': 0.784, 'recommendation': 'time_passed'}, '08-09': {'avg_house_consumption': None, 'solcast_pv_estimate': 2.193, 'estimated_net_consumption': None, 'estimated_cost': 0.0, 'batteries_charged': 0.0, 'import_price': 2.204, 'export_price': 0.772, 'recommendation': 'time_passed'}, '09-10': {'avg_house_consumption': None, 'solcast_pv_estimate': 3.383, 'estimated_net_consumption': None, 'estimated_cost': 0.0, 'batteries_charged': 0.0, 'import_price': 2.089, 'export_price': 0.68, 'recommendation': 'time_passed'}, '10-11': {'avg_house_consumption': None, 'solcast_pv_estimate': 4.41, 'estimated_net_consumption': None, 'estimated_cost': 0.0, 'batteries_charged': 0.0, 'import_price': 2.049, 'export_price': 0.648, 'recommendation': 'ev_smart_charging'}, '11-12': {'avg_house_consumption': None, 'solcast_pv_estimate': 5.14, 'estimated_net_consumption': None, 'estimated_cost': 0.0, 'batteries_charged': 0.0, 'import_price': 1.852, 'export_price': 0.491, 'recommendation': 'batteries_charge_solar'}, '12-13': {'avg_house_consumption': None, 'solcast_pv_estimate': 5.659, 'estimated_net_consumption': None, 'estimated_cost': 0.0, 'batteries_charged': 0.0, 'import_price': 1.629, 'export_price': 0.312, 'recommendation': 'batteries_charge_solar'}, '13-14': {'avg_house_consumption': None, 'solcast_pv_estimate': 5.747, 'estimated_net_consumption': None, 'estimated_cost': 0.0, 'batteries_charged': 0.0, 'import_price': 1.476, 'export_price': 0.19, 'recommendation': 'batteries_charge_solar'},  [... cut ...]
2025-08-13 10:19:31,462 - DEBUG - Hour: 23-00 | Summer: no solar estimate, setting recommendation to BatteriesDischargeMode. | Import Price: 2.207 | Export Price: 0.775 | Net Consumption: None
2025-08-13 10:19:31,462 - DEBUG - Completed optimization strategy for all remaining hours.
2025-08-13 10:19:31,462 - DEBUG - # Recommendation for 10-11 is EV Charger is active. Setting TOU Periods: None and Working Mode: maximise_self_consumption, House Power: -88, Batteries Discharge Power: 50
2025-08-13 10:19:31,462 - DEBUG - ------ Completed updating working mode sensor state...

jbruntt, 2025-08-13:
I did a new test. Charging the car with higher wattage than  what my panels are producing.
HSEM can see that the car is charging (sensor.hsem_workingmode_sensor/**ev_charger_status_state: True**)
HSEM says that the battery is in maximise_self_consumption (sensor.hsem_workingmode_sensor/huawei_solar_batteries_working_mode_state: **maximise_self_consumption**)
The battery is discharging at about 2900W.
Here is what I have collected.
### Debug-information from selected sensors
=== SENSOR DATA ===
Battery Working Mode (Huawei Direct): maximise_self_consumption
Sensor: select.batteries_working_mode

HSEM Working Mode: ev_smart_charging
Sensor: sensor.hsem_workingmode_sensor

EV Charger Status State (HSEM): True
Attribute: sensor.hsem_workingmode_sensor -> ev_charger_status_state

Battery Working Mode (HSEM View): maximise_self_consumption
Attribute: sensor.hsem_workingmode_sensor -> huawei_solar_batteries_working_mode_state

EV Status (Home Assistant): charging
Sensor: sensor.carport_jens_bruntt_status

EV Power (Home Assistant): 7.56699991226196W
Sensor: sensor.carport_jens_bruntt_power

=== BATTERY OPTIONS ===
Available Battery Working Modes: ['adaptive', 'fixed_charge_discharge', 'maximise_self_consumption', 'fully_fed_to_grid', 'time_of_use_luna2000']

=== POWER STATUS ===
Battery Discharge Power: 2779.0W
Battery Charge/Discharge Power: -2779W
Solar Production: 5735.0W
Battery SOC: 74.0%

=== ALLE HSEM WORKING MODE ATTRIBUTES ===
batteries_conversion_loss: 10.0
batteries_current_capacity: 7.2
batteries_usable_capacity: 9.5
energi_data_service_export_entity: sensor.energi_data_service_sale
energi_data_service_export_min_price: -0.22
energi_data_service_export_state: 0.293
energi_data_service_import_entity: sensor.energi_data_service
energi_data_service_import_state: 1.605
energy_needs: {'0am_6am': 0.0, '6am_10am': 0.0, '10am_5pm': 0.0, '5pm_9pm': 0.0, '9pm_midnight': 0.0}
ev_charger_force_max_discharge_power: False
ev_charger_max_discharge_power_state: 500
ev_charger_power_entity: sensor.carport_jens_bruntt_power
ev_charger_power_state: 7.57
ev_charger_status_entity: sensor.carport_jens_bruntt_status
ev_charger_status_state: True
force_working_mode_entity: select.huawei_solar_energy_management_force_working_mode
force_working_mode_state: auto
hourly_calculations: {'00-01': {'avg_house_consumption': None, 'solcast_pv_estimate': 0.0, 'estimated_net_consumption': None, 'estimated_cost': 0.0, 'batteries_charged': 0.0, 'import_price': 2.06, 'export_price': 0.696, 'recommendation': 'time_passed'}, '01-02': {'avg_house_consumption': None, 'solcast_pv_estimate': 0.0, 'estimated_net_consumption': None, 'estimated_cost': 0.0, 'batteries_charged': 0.0, 'import_price': 2.006, 'export_price': 0.653, 'recommendation': 'time_passed'}, '02-03': {'avg_house_consumption': None, 'solcast_pv_estimate': 0.0, 'estimated_net_consumption': None, 'estimated_cost': 0.0, 'batteries_charged': 0.0, 'import_price': 1.977, 'export_price': 0.63, 'recommendation': 'time_passed'}, '03-04': {'avg_house_consumption': None, 'solcast_pv_estimate': 0.0, 'estimated_net_consumption': None, 'estimated_cost': 0.0, 'batteries_charged': 0.0, 'import_price': 1.957, 'export_price': 0.614, 'recommendation': 'time_passed'}, '04-05': {'avg_house_consumption': None, 'solcast_pv_estimate': 0.0, 'estimated_net_consumption': None, 'estimated_cost': 0.0, 'batteries_charged': 0.0, 'import_price': 1.979, 'export_price': 0.632, 'recommendation': 'time_passed'}, '05-06': {'avg_house_consumption': None, 'solcast_pv_estimate': 0.005, 'estimated_net_consumption': None, 'estimated_cost': 0.0, 'batteries_charged': 0.0, 'import_price': 2.037, 'export_price': 0.678, 'recommendation': 'time_passed'}, '06-07': {'avg_house_consumption': None, 'solcast_pv_estimate': 0.228, 'estimated_net_consumption': None, 'estimated_cost': 0.0, 'batteries_charged': 0.0, 'import_price': 2.232, 'export_price': 0.794, 'recommendation': 'time_passed'}, '07-08': {'avg_house_consumption': None, 'solcast_pv_estimate': 0.915, 'estimated_net_consumption': None, 'estimated_cost': 0.0, 'batteries_charged': 0.0, 'import_price': 2.219, 'export_price': 0.784, 'recommendation': 'time_passed'}, '08-09': {'avg_house_consumption': None, 'solcast_pv_estimate': 2.193, 'estimated_net_consumption': None, 'estimated_cost': 0.0, 'batteries_charged': 0.0, 'import_price': 2.204, 'export_price': 0.772, 'recommendation': 'time_passed'}, '09-10': {'avg_house_consumption': None, 'solcast_pv_estimate': 3.383, 'estimated_net_consumption': None, 'estimated_cost': 0.0, 'batteries_charged': 0.0, 'import_price': 2.089, 'export_price': 0.68, 'recommendation': 'time_passed'}, '10-11': {'avg_house_consumption': None, 'solcast_pv_estimate': 4.491, 'estimated_net_consumption': None, 'estimated_cost': 0.0, 'batteries_charged': 0.0, 'import_price': 2.049, 'export_price': 0.648, 'recommendation': 'time_passed'}, '11-12': {'avg_house_consumption': None, 'solcast_pv_estimate': 5.222, 'estimated_net_consumption': None, 'estimated_cost': 0.0, 'batteries_charged': 0.0, 'import_price': 1.852, 'export_price': 0.491, 'recommendation': 'time_passed'}, '12-13': {'avg_house_consumption': None, 'solcast_pv_estimate': 5.676, 'estimated_net_consumption': None, 'estimated_cost': 0.0, 'batteries_charged': 0.0, 'import_price': 1.629, 'export_price': 0.312, 'recommendation': 'time_passed'}, '13-14': {'avg_house_consumption': None, 'solcast_pv_estimate': 5.882, 'estimated_net_consumption': None, 'estimated_cost': 0.0, 'batteries_charged': 0.0,  [... cut ...]
2025-08-13 14:22:50,124 - DEBUG - # Recommendation for 14-15 is EV Charger is active. Setting TOU Periods: None and Working Mode: maximise_self_consumption, House Power: 2831, Batteries Discharge Power: 2850
2025-08-13 14:22:50,124 - DEBUG - ------ Completed updating working mode sensor state...
2025-08-13 14:22:54,222 - DEBUG - # Recommendation for 14-15 is EV Charger is active. Setting TOU Periods: None and Working Mode: maximise_self_consumption, House Power: 2831, Batteries Discharge Power: 2850
2025-08-13 14:22:54,222 - DEBUG - ------ Completed updating working mode sensor state...

woopstar, 2025-08-13:
2025-08-13 14:22:54,222 - DEBUG - # Recommendation for 14-15 is EV Charger is active. Setting TOU Periods: None and Working Mode: maximise_self_consumption, House Power: 2831, Batteries Discharge Power: 2850

It now set to 2850. Now I remember. We introduced in 3.2 that it will discharge to cover the house load, but not the charger load. So the battery will discharge the same as your house load.

I can make this optional. It was requested in another issue. I think it was @Extati

jbruntt, 2025-08-13:
I don't think that my house consumption at the time was 2900w, but I might be wrong.
The configuration @Extati and @woopstar are talking about is this one - below. Note that I have it disabled. And the 500W configuration was because I was hoping HSEM would pick it up. 
I agree that the batteries should stop discharging when the HSEM is operating in "EV Smart Charging" mode if the Force Max is disabled.

[image]

woopstar, 2025-08-13:
> batteries should stop discharging when the HSEM is operating in "EV Smart Charging" mode if the Force Max is disabled

Then we have found the issue. To mitigate this while I fix it, you can enable Force Max Discharge Power and set it to 50W.

jbruntt, 2025-08-14:
> > batteries should stop discharging when the HSEM is operating in "EV Smart Charging" mode if the Force Max is disabled
> 
> Then we have found the issue. To mitigate this while I fix it, you can enable Force Max Discharge Power and set it to 50W.

Looks like this workaround works for me. 
Thanks

closed 2025-08-14 by jbruntt