woopstar/hsem #150: "Huawei Batteries Maximum Discharging Power (W)" - Exceeding Battery's capabilities

madmat17, 2025-08-17:
I'm currently testing the HSEM, which seems to be very promising (as today I'm controlling my battery's behavior via complex automations, which are not always as reliable as needed).

However, I stumbled across the following bug/problem:
According to the log files the HSEM integration seems to be setting the maximum discharge power of the battery to **5000 W**. The only problem is, that not all batteries are capable of charging/discharging that much power.
The `Huawei Solar` integration is paying respect to that by limiting the respective number sensor's range accordingly (based on the maximum power capabilities according to the respective modbus register).
e.g.: The Luna2000 with 5 kWh capacity can only charge/discharge a power of **2500 W**.

That led to the problem, that the integration couldn't update the maximum discharge sensor...
```
Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/helpers/entity_platform.py", line 676, in _async_add_entities
    await self._async_add_entity(
        entity, False, entity_registry, config_subentry_id
    )
  File "/usr/src/homeassistant/homeassistant/helpers/entity_platform.py", line 1002, in _async_add_entity
    await entity.add_to_platform_finish()
  File "/usr/src/homeassistant/homeassistant/helpers/entity.py", line 1380, in add_to_platform_finish
    await self.async_added_to_hass()
  File "/config/custom_components/hsem/custom_sensors/working_mode_sensor.py", line 2506, in async_added_to_hass
    await self._async_handle_update(None)
  File "/config/custom_components/hsem/custom_sensors/working_mode_sensor.py", line 641, in _async_handle_update
    await self._async_set_working_mode()
  File "/config/custom_components/hsem/custom_sensors/working_mode_sensor.py", line 1042, in _async_set_working_mode
    await async_set_number_value(
    ...<3 lines>...
    )
  File "/config/custom_components/hsem/utils/misc.py", line 222, in async_set_number_value
    await self.hass.services.async_call(
    ...<7 lines>...
    )
  File "/usr/src/homeassistant/homeassistant/core.py", line 2835, in async_call
    response_data = await coro
                    ^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/core.py", line 2878, in _execute_service
    return await target(service_call)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/helpers/service.py", line 850, in entity_service_call
    single_response = await _handle_entity_call(
                      ^^^^^^^^^^^^^^^^^^^^^^^^^^
        hass, entity, func, data, call.context
        ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
    )
    ^
  File "/usr/src/homeassistant/homeassistant/helpers/service.py", line 922, in _handle_entity_call
    result = await task
             ^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/components/number/__init__.py", line 106, in async_set_value
    raise ServiceValidationError(
    ...<8 lines>...
    )
homeassistant.exceptions.ServiceValidationError: Value 5000.0 for number.battery_maximale_entladeleistung is outside valid range 0.0 - 2500
```

...and subsequently also couldn't initialize the working mode sensor.
[image]

This is being caused by the 5000 W to be hardcoded:
https://github.com/woopstar/hsem/blob/04495d16404de2577e6c473e669cfe738e8c54d9/custom_components/hsem/custom_sensors/working_mode_sensor.py#L1036-L1046

I assume that the hardcoded value of **5000 W** will also be used for charging / discharging time calculations, right?
It might be a better approach to read the maximum charge and discharge values (e.g. by checking the sensor attributes).
Let me know, in case additional logs/tests would be of any help.

Cheers
Mat

woopstar, 2025-08-17:
Hi Mat,

Thank you! I will correct the issue in the next update.

jonaslevring, 2025-08-26:
it says 9,5 - but how come that ? - is this a calculation and not absolute number.. 10000/1000 should not be 9,5 :-)

woopstar, 2025-08-26:
> Huawei solar batteries maximum discharging power state

Maximum discharge power is 5 kW for +10 kW systems and 2,5kW for less. So this is correct.

I forgot, we have added a small buffer to the max capacity back in the days, so you are seeing 9,5 due to that buffer. I can remove the buffer as it does not do much more. Will release a fix later today for both issues.

jonaslevring, 2025-08-26:
Thank you.. looking forward to.. will check next time if it behaves correctly.. i had an observation that it kept on draining my battery despite in EV mode and also trying to set max discharge to 50W.. 
(I would like not to use any battery while charging car, as charging already is scheduled to the cheapest time, and thereby i would like to avoid degration on house battery)
But will check into if thats still and issue afterwards.